imchipwood/dumbpad #1: LED Question

revere521, 2020-02-21:
Hello sir - I was wondering if you could help me understand what code to modify in my keymap.c to change the behavior of one of the LEDs?

The LED that lights when the layer key is pressed (or toggled) works fine, but the other never illuminates.   I'd like to see if I could set it to light when i press the center button on the rotary encoder (currently set to mute on the main layer) 

I see the section below in the keymap.c, but its not clear to me how to leverage it

```
void led_set_user(uint8_t usb_led) {
}
```

imchipwood, 2020-04-15:
Woah sorry I never saw this until now! I have not really tested the LED code - I only just ordered some new boards with the LED sockets last weekend, they haven't arrived yet.

If I wrote the code correctly, both LEDs should be used to indicate the layer in a binary fashion. 

Assuming the right LED is the 1s position and the left LED is the 2s position, it should look like this
- layer 0: 00 (both LEDs off)
- layer 1: 01 (right LED on)
- layer 2: 10 (left LED on)
- layer 3: 11 (both LEDs on)

If you don't ever use layer 2 or higher, the left LED shouldn't turn on unless you add custom code to `led_set_user`

Look in `dumbpad.c` `layer_state_set_kb` to see how the LEDs are turned on/off - you can change what the LEDs do in your `led_set_user` function by calling `writePin` on the LED of interest (`LAYER_INDICATION_LED_0` or `LAYER_INDICATION_LED_0`) with a value of 0 (off) or 1 (on). How you decide when to turn an LED on or off is up to you.

revere521, 2020-04-15:
no worries at all.  I think i found this through trial and error - since I'm only using 1 layer, the second light has no function in the current code.

I also designed some 3d printed cases for your board if you are interested: https://www.thingiverse.com/thing:4175635

imchipwood, 2020-04-15:
Awesome, hope you found a use for that LED.

I'm closing this issue since you've got another one open about the case.

closed 2020-04-15 by imchipwood