imchipwood/dumbpad #2: Not An Issue - I Designed Some 3D Printable Cases if you are interested

revere521, 2020-02-22:
I designed these if you are interested - nice work by the way - the PCB's are great - I've had some fun with this

https://www.thingiverse.com/thing:4175635

imchipwood, 2020-04-15:
Wow, sorry I missed this! Not sure why I didn't get notified of the issue - will look into that...

These are super cool - I like that you wired the rotary encoder as an external component for use with that huge knob!

Do you mind if I the thingiverse link to the READMEs?

revere521, 2020-04-15:
I do not mind at all, please do!

I did make a smaller case too with the encoder on the board, but with the board still upside down.  The STEP files are there so it can be modified as a starting point fo rother configurations

closed 2021-12-08 by imchipwood