imchipwood/dumbpad #3: hi, i had an issue on the encoder and the third led

ryandiwdjj, 2020-06-21:
the default mapping on qmk_firmware, the rotary and the third LED doesn't work (im using the v1.1 board) but i already make the fix and if you want i can send my files. (sorry for my bad english as it is not my mother language)

imchipwood, 2020-06-22:
What branch did you use to build the qmk_firmware? The correct branch is `dumbpad_refactor` - maybe I forgot to push some code?

ryandiwdjj, 2020-06-23:
i dont see any branch named dumbpad_refactor

imchipwood, 2020-06-23:
OK, I see. You're talking about this repo and the Eagle files.

No update is need in this repository if you use the correct qmk_firmware branch.

I am using a v1.1 right now and it's working fine with the code in [imchipwood/qmk_firmware/tree/dumbpad_refactor](http://github.com/imchipwood/qmk_firmware/tree/dumbpad_refactor)

imchipwood, 2020-06-23:
The reason for this: v1.1 is significantly different from v1.0, in order to support both single- and dual-encoder layouts. The v1.1 layout uses the same encoder & LED pins as v0.6_dualencoder - the v1.1 design is actually based around the original dualencoder designs.

ryandiwdjj, 2020-06-23:
aahh i see, anyway i have already fix it by myself after i trace it one by one LOL, i just though maybe someone out there had the same issue as me, thanks for the reply though! the design sure is beautiful!

imchipwood, 2020-06-23:
Glad you like it, do you mind sending me a photo of yours? I like seeing what people have made :)

ryandiwdjj, 2020-06-24:
here's mine, i use the third LED as my third layer indicator lmao
https://imgur.com/1YVRIGx

imchipwood, 2020-06-24:
Awesome, I hadn't seen anyone put the encoders in opposite corners like that yet! 

Thanks for the photo - do you mind if I include it in the v1.1 README?

ryandiwdjj, 2020-06-24:
yea i don't mind, hahahaha i think its kinda weird too but that is the most comfortable layout for me, also i'm going to try to make the acrylic case for it in free time, when its done i'm going to put the file in this issue

imchipwood, 2020-06-24:
I'd actually like to close this issue - feel free to submit another issue when you've made the case!

closed 2020-06-24 by imchipwood