imchipwood/dumbpad #4: Flash mode

shaye1444, 2021-01-01:
How would one go into flash mode to change firmware?

breakawayt, 2021-01-01:
> How would one go into flash mode to change firmware?

Two methods that i've discovered:
1. Double-press the switch button (top of rotary encoder, if you've got this soldered)
2. Hold down rotary encoder pushbutton while plugging in to enter bootloader (according to config.h)
[image]

imchipwood, 2021-01-04:
If you're simply looking to flash, you only have to press the reset button once. This resets the Arduino and will temporarily enter the bootloader during the normal boot sequence - if your computer is currently attempting to flash the board, it will detect this and flashing will begin. Double pressing the reset button will enter the bootloader for 8 seconds.

These methods will work if you've already flashed the Arduino with QMK once:
* add `RESET` keycode to your keymap - this will also force it to enter the bootloader for 8 seconds
* hold down rotary encoder button while plugging in USB cable (or hit reset button once) to enter bootloader

closed 2021-01-04 by imchipwood