imchipwood/dumbpad #7: Issue trying to print the gerbers at jlcpcb

zpeleg, 2021-03-11:
Hi, I tried producing the combo gerbers and got a response mail from jlcpcb and I'm not sure what they mean. They ask:

"As shown below, shall we make these indicated slots as plated slots, or the tin bath slots?"

What do they mean? What do I answer?

[image]

ephphatha, 2021-03-23:
It looks like they're indicating the ovalised slots used to anchor rotary encoders. The plating for those slots is for structural reasons (lets you solder some anchor tabs to hold the rotary encoders better). I'm not 100% sure what the different options they're talking about are but I suspect "tin bath slots" refers to making through hole plated slots and then treating them with a tinning solution to make them easier to solder, whereas "plated slots" would skip this tinning step. Since they're not electrical connections it doesn't really matter how they're plated so I'd go with whatever option is cheaper.

zpeleg, 2021-03-23:
Sorry for not updating! I thought that this is what they meant as well, but it's not. Apparently it's simply a bad translation and they meant either plated though holes or non plated through hole. I went with plated.

Just got the pcbs and waiting for the rest of the components. Will update with photos ;)

closed 2021-03-23 by zpeleg

imchipwood, 2021-03-23:
Could've sworn I responded to this - but yes, these are plated slots for the ground/chassis pins of the encoder. You were correct to request they plate them!