imchipwood/dumbpad #8: 1.1 - QMK no device detected

Aldarith, 2021-04-25:
Hey Folks,

trying to flash some fresh firmware on my dumbpad.

No matter what I do, whenever I click 'Flash', QMK tells me no device is detected.

I have hit the reset ground pin, attempted to hold the ground pin while inserting cable, etc....

imchipwood, 2021-04-25:
Do any of the LEDs on the pro micro turn on? Can you program it using the Arduino IDE?

It sounds to me like your Pro Micro may be damaged, but it's hard to say without pictures of your solder job.

Aldarith, 2021-04-26:
Hey! Thanks for the quick response.

Yeah, it works fine! I flashed it once before but I forgot to add a '0' key so now I am attempting to correct that mistake.
All the buttons work as intended so far.

Aldarith, 2021-04-26:
[image]

I am sure I just have this set up wrong somehow...

imchipwood, 2021-04-26:
MCU seems correct, though last time I programmed any QMK boards from Windows I used dfu_util, not avrdude. That was almost a year ago now so maybe QMK changed which programmer QMK Toolbox uses.

My only guess is permissions WRT the com port - could try running QMK Toolbox as admin. If that doesn't work, you might want to take this to the QMK issue system or discord - it doesn't appear to specifically be a dumbpad problem.

imchipwood, 2021-12-08:
Closing this issue - please reopen if you're still experiencing this problem

closed 2021-12-08 by imchipwood