imchipwood/dumbpad #9: I'm a newbie

shaye1444, 2021-05-17:
I'm a newbie,

I was able to build my dumpad and everything then I tried to flash with QMK and now none of the keys work :( I'm really not sure what to do as I am not versed in programming. Any help for me?

Thanks!

imchipwood, 2021-05-17:
Hey - sorry to hear you're having issues, but I'm happy to help out! 

Few questions for ya:
* What PCB version? 1.1 or something else?
* What layout? i.e. 1 encoder in the bottom left, dual encoder, etc.
* What operating system is your computer? Windows, macOS, etc.
* How did you build the firmware and flash it? 
  * In case you built it yourself, you can now use [QMK Configurator](https://config.qmk.fm/#/dumbpad/v1x/LAYOUT) to build your firmware if you're not comfortable with programming.

imchipwood, 2021-12-08:
Closing issue due to lack of response. Please reopen if you still need assistance!

closed 2021-12-08 by imchipwood