imchipwood/dumbpad #13: Dumbpad issue

tarnis, 2021-11-28:
probably is a better way to test the pcb before I started but I'm not 100% sure what to touch to what on the voltmeter :)

my buttons seem to be working touching where I soldered, but these ones aren't functioning still.  Wasn't sure if there was a common point maybe on the pro micro I might need to resolder?

[image]

imchipwood, 2021-11-28:
Rows each have their own pin, and columns each have their own pin. The row you've circled is ROW2 which is connected to A1.

The other switch you've circled (row 3, column 3), since the problem is isolated to a single switch, it's probably the diode. Check its polarity.

tarnis, 2021-11-30:
Thanks, a little more solder on both fixxed it...good thing as I can't remove that plastic mount really without removing all of the switches.

If you don't mind my asking, where would I look to see how things are assigned to each pin?  Eventually I think I'm looking to make a controller for a car I built with something like this and making my own pcb.

Thanks,
Matt

imchipwood, 2021-11-30:
Woo! Glad you got it working! 

Look at the .brd (board) and .sch (schematic) files using Eagle. Columns are ordered 0-4 from left to right, where column 0 is just the single encoder under the pro micro. Rows are ordered 0-3 from top to bottom.

If you click a wire in the board file and then click the 'i' information button it'll give you the name of the wire (ROW0, COL3, etc.). Switch over to the schematic and you can see which pin on the pro micro that connects to.

closed 2021-12-08 by imchipwood