imchipwood/dumbpad #15: Change VENDOR_ID in QMK firmware for proper VIA compatibility

mpfl, 2022-01-25:
Theoretically this is an issue to raise in the QMK repo, but I wanted to get your attention.

It looks like the VENDOR_ID for the Dumbpad is still the default of 0xFEED.

Would you be able to change the VENDOR_ID so that I can submit my VIA config to VIA? Alternatively, I would be happy to change the VENDOR_ID and submit a PR. Figure it would be polite to ask you first since you're the Dumbpad's creator!

imchipwood, 2022-01-25:
Huh, thought I'd changed it to DEAF or something else silly.. Anyway thanks
for contacting me first - I don't really care what the vendor ID is so long
as the device ID still shows up as 'dumbpad' when plugged in!

Feel free to change it!

medoix, 2022-08-05:
Just noting that my PR to merge in the OLED firmware also makes the ```VENDOR_ID``` change here https://github.com/qmk/qmk_firmware/pull/17058/files#diff-c34bacd1d7271500b70bcbec925dae2da916c0e295de920decee8f4e66a67d78R22

This will then unblock the one current reason for no VIA support in this PR https://github.com/the-via/keyboards/pull/1255