imchipwood/dumbpad #16: Mill Max compatible?

deveth0, 2022-01-28:
Hey, thank you so much for the board layouts!
I plan to build a "hotswap" able version, do you know by any chance, if the board i compatible to Mill Max sockets? 

Cheers

imchipwood, 2022-01-28:
Never heard of them - unless they have the same pin locations as a cherry
MX switch, they won't be compatible.

deveth0, 2022-01-28:
oh wow, that's fast, thanks! 
Mill max are just soldered into the wholes and provide some kind of hotswap for all kind of switches:
https://keygem.store/products/mill-max-hot-swap-sockets?_pos=1&_sid=fd8fb5200&_ss=r

imchipwood, 2022-01-28:
Hmm, looks like it should work. I can't remember exactly but the holes should be the standard 0.060" (1.524mm). [This page](https://www.mill-max.com/search/tags/hot--swap-socket?keyword=keyboard%20switch%20socket&creative=&msclkid=4cba2cd1ad3f1726e10f4d25bdd1b5a5) shows the sleeve outer diameter at just under that (0.058" (1.47mm)), so they should squeeze in there.

Let me know if you try it out - I'm not much of a hot swapper but I've never seen these before and am curious how well they hold onto a switch!

deveth0, 2022-02-08:
Sorry, can't provide feedback here as I did not order your PCB  :(