imchipwood/dumbpad #23: hot swap, rgb and oled

ykz89, 2022-05-10:
Hi,

Now that there is a hotswap/rgb version and an oled version, is there any chance you could create a version with all of those features?

imchipwood, 2022-05-11:
[That idea has actually already been floated](https://github.com/imchipwood/dumbpad/pull/20#issuecomment-1109500930) :)

But, at the moment, neither the OLED or RGB versions are supported by the QMK master branch - therefore do not work with VIA or the QMK configurator out-of-the-box... [There is a PR](https://github.com/qmk/qmk_firmware/pull/16285) to get the RGB version supported, I'll apply some pressure to get that in...

@medoix what changes were made to QMK support the OLED version? Could they be merged into QMK's master branch?

ykz89, 2022-05-11:
> [That idea has actually already been floated](https://github.com/imchipwood/dumbpad/pull/20#issuecomment-1109500930) :)
>
> But, at the moment, neither the OLED or RGB versions are supported by the QMK [...]

Is it perhaps an idea to include the changes needed, in this repository until it is merged into QMK?

medoix, 2022-05-11:
@imchipwood i have created a [PR for VIA](https://github.com/the-via/keyboards/pull/1255) support of the dumbpad in general, the OLED version has the same switch/encoder layout so would be configurable by VIA the same.

I have just raised a [QMK PR](https://github.com/qmk/qmk_firmware/pull/17058) for the OLED version.

imchipwood, 2022-05-11:
@ykz89 I don't really want to keep any code in this repo, and it looks like all of the QMK changes needed for OLED + RGB are in PR so I don't have to! If we get a PCB design finished before the QMK PRs go thru we will certainly add a link to the appropriate branches in the README.

I'll look into making an rgb+hotswap+oled version this weekend - I need to learn KiCad anyway and this is a small enough change to the RGB version it should be a good learning experience.

bagges, 2022-06-09:
@imchipwood How's your KiCad learning going? :-)

imchipwood, 2022-06-09:
> @imchipwood How's your KiCad learning going? :-)

Well, I installed KiCad on my desktop... :P

TimoWielink, 2022-08-21:
Happy to Contribute to creating this version, But I am struggling with the use of Both KiCad and Eagle. Happy to collaborate to come up with a better system for the different boards (And also work on the naming?).

@imchipwood Did you manage to work on an Oled + RGB + Hotswap version, seems like this has been open for a while. Very happy to pick up any work and support the project.

imchipwood, 2022-08-21:
Unfortunately, no work has been done on this... Too busy with summer stuff!

In all reality I likely won't work on any personal projects until the
middle of next year due an 8 month class/training that starts soon..

TimoWielink, 2022-08-22:
No worries, I will see what I can do! I just ordered a batch of PCBs and will see how they turn out. 

p.s. Not related to this issue, but is there a space where roadmap, features etc are being discussed for the dumbpad?

imchipwood, 2022-08-22:
Nope, no plans or place to discuss them... Open to suggestions, though!

TimoWielink, 2022-08-24:
@imchipwood Could be interested to spin up a Discord or Slack. I can give it some thought over the weekend and share my progress on my build and plans for it.

Thnx

dickofn, 2022-10-11:
I'm bumping the old PR for this and following up with the reviewers

all credit goes to @imchipwood and @deveth0 , if you don't like my action, please let me know, and I'll take down the PR
this PR is essentially your PR with modifications as requested by the reviewer before the PR is closed

dickofn, 2022-10-11:
**New Update for V3X**

I've just submitted a PR to Vial https://github.com/vial-kb/vial-qmk/pull/261
it includes rotary encoder support and RGB matrix with VIAL software

It's been hard having no official support, but now it'll be over soon cheers 🥂 

Credit goes to:
Board @imchipwood
QMK for v3x @deveth0

imchipwood, 2022-10-11:
Thanks for getting via/vial working! Hopefully the PR goes through soon and we can link users to the various GUIs they can build firmware with.

dickofn, 2022-10-11:
you're welcome sir, I need this device to be more user-friendly because I want to gift my other dumbpad to less techy guys

dickofn, 2022-10-12:
It's finally merged to vial master https://github.com/vial-kb/vial-qmk

dickofn, 2022-10-13:
And also in qmk master, cheers 🥂 
https://github.com/qmk/qmk_firmware

nuddler, 2025-02-04:
Is it possible to get combo_oled with hot_swap ?

imchipwood, 2025-02-05:
> Is it possible to get combo_oled with hot_swap ?

I have no plans to work on dumbpad anymore - any updates will have to come from the community. Feel free to design it yourself :)