imchipwood/dumbpad #27: Type C pro micro compatibility

pacjo, 2022-10-08:
Hi, I'm thinking about building the `hotswap_rgb` version, but I want to have usb c connection.

I found this: https://pl.aliexpress.com/item/32840365436.html and I wanted to use the `TYPE-C USB 3-6V NEW` and I was wondering if it will work with the `hotswap_rgb`.

I was wondering if you (or anyone else) has any experience with USB C pro micros. If they are already tested (they work or not) would you mind adding this info to the readme?

bubbernail, 2022-10-08:
Hi. I built one myself with this controller and I have zero issues. It's great.

pacjo, 2022-10-08:
> Hi. I built one myself with this controller and I have zero issues. It's great.

that's very good to hear