imchipwood/dumbpad #31: Dumbpad trouble

Hintaa, 2022-11-28:
[image]
I just finished flashing my dumbpad but the two circled columns wont work

imchipwood, 2022-11-28:
Check your soldering on all the pins for those switches, the diodes next to them, and the pins on the arduino for those columns. Double check diodes are the correct direction. 

If you have a multimeter, you can check the voltage on the column pins while pressing buttons. It should change, not 100% sure what values you should see and not currently at home to double check.

Hintaa, 2022-11-28:
Which pins on the arduino should they be?
should it be the 8 &9 on the left side pf the arduino

imchipwood, 2023-02-02:
Sorry for the long silence - are you still having issues? Would need to sit down and look things over to verify pin connections.

Hintaa, 2023-02-03:
Yes I am still having trouble

Hintaa, 2023-02-03:
oop sorry accident

imchipwood, 2025-02-05:
@Hintaa are you still having issues? I would like to close this if you resolved the issue.