imchipwood/dumbpad #34: Dumbpad not working with any firmware I got from QMK configurator

TylerWang2, 2023-03-27:
My dumbpad PCB version is 1.1, 17 keys, encoder on the bottom left, with an OLED display

The keys itself is working as I found some random firmware to flash it and it is working. As soon as i use one of my own configuration from QMK configurator non of the keys work

imchipwood, 2023-03-27:
17 keys with encoder on the bottom left? Do you mean 16 keys?

Export your QMK configurator config as JSON and paste it into here - I'll try to look at it today or tomorrow to see if I can reproduce the problem.

TylerWang2, 2023-03-28:
sorry, yes.
Here is the JSON:
{
    "name": "DUMBPAD V1.1",
    "vendorId": "0xA153",
    "productId": "0xF3BE",
    "lighting": "none",
    "matrix": { "rows": 4, "cols": 5 },
    "layouts": {
      "keymap": [
        [{"x": 1.25}, "0,1", "0,2", "0,3", "0,4"],
        [{"x": 1.25}, "1,1", "1,2", "1,3", "1,4"],
        [{"x": 1.25}, "2,1", "2,2", "2,3", "2,4"],
        ["3,0", {"x": 0.25}, "3,1", "3,2", "3,3", "3,4"]
      ]
    }
  }

imchipwood, 2023-04-16:
Sorry I took so long to get back. 

I didn't notice you mentioned the OLED version in your original issue - I actually cannot help you much with that one as I do not have an OLED version (I did not design that one). You'll want to talk to the guy behind keebd: https://github.com/keebd/dumbpad

closed 2023-04-16 by imchipwood