imchipwood/dumbpad #35: Troubleshooting help

afishydeath, 2023-04-16:
Hi there, I recently purchased the oled version from keebd, and after soldering it all together today, only the reset button and the knob are recognized. I have tried continuity tests from the pins of the pro micro through the switches, and using diode mode on my multimeter everything looks like expected, however after multiple attempts with different versions of firmware (starting with the ones on keebd's website, I still get 0 output from the switches in tests. Any advice on how to troubleshoot would be appreciated.

Slightly more info on continuity testing. I specifically checked the top left key, the row pin is f4 and the column pin is e6, i get continuity when pressed, but still nothing in a keyboard test, using suggested firmware

imchipwood, 2023-04-16:
So, keyboards measure keypresses in a row/column matrix. A single key press causes a voltage drop across a diode on two pins - a row pin, and a column pin. Multiple keypresses compound this - by measuring the voltage on each pin of the matrix, you can determine which 

Therefore, measuring the continuity from _one_ pin doesn't really tell you much of anything. In order for a single key to work, continuity must be good between each key, diode, _and_ microcontroller pin for the entire row and column.

I would open the schematic for the combo_oled board (download KiCAD), and check continuity between every single connection.

Or just resolder every single diode/switch/microcontroller pin..

afishydeath, 2023-04-16:
Okay, I'll give that a try.

imchipwood, 2023-07-27:
Closing issue due to no updates in a few months. Please reopen if you're still having issues.

closed 2023-07-27 by imchipwood