imchipwood/dumbpad #36: Dumbpad issue...switches not working

tarnis, 2023-06-30:
Hi, building a second dumbpad tonight, the column next to the rotory encoder slot and led's isn't working.  What is that linked to?

Great job on this board design btw!

imchipwood, 2023-06-30:
Which version did you build?
-Chip

tarnis, 2023-06-30:
Promicro

imchipwood, 2023-06-30:
They all use a pro micro... Rgb? Oled? No LEDs?

tarnis, 2023-07-01:
Lol...just the 3 LEDs above the rotary encoder.  Not a fancy one.  Combo I guess.  😁. Made these a long time ago.

imchipwood, 2023-07-01:
Column 1 is pin D6 on the Pro Micro

Columns are numbered 0-4 from left to right, rows are 0-3 from top to bottom
[image: image.png]
-Chip

imchipwood, 2023-07-01:
Image failed to send via email...
[image]

tarnis, 2023-07-01:
That did the trick, appreciate your help!  Wish these worked with the hot swap sockets I bought....case looks like a trainwreck now 😞

closed 2023-07-01 by tarnis