imchipwood/dumbpad #37: Dual Encoders Tied Together

tarnis, 2023-07-10:
I'm finally getting around to setting up and using the dumbpad I made last year and put a ticket in on a year ago :)  There's a photo in it if it helps.

Seems like when I turn the left encoder of my dual encoder setup using the default profile keymap for v1x_dualencoder it adjusts the volume and moves the mouse left/right simultaneously.

imchipwood, 2023-07-10:
I've never built one of the dual encoders, so I can't say their profiles
have ever actually been tested...

If you make a new profile and fix the issue, please send it to me and I'll
use it to fix whatever is wrong with the default profile.

imchipwood, 2025-02-05:
No response in over a year, closing issue. Please open a new issue if any new findings are available.

closed 2025-02-05 by imchipwood