imchipwood/dumbpad #39: it simply doesn't work

csmayen, 2023-07-30:
Hello, when I connect the Pro micro to QMK Toolbox recognizes it perfectly as a dumbpad but when I place it on the keyboard it simply doesn't work, not even the reset button. All the soldering seems correct.

[image]
[image]

imchipwood, 2023-07-30:
Sounds like something is shorted, probably reset to ground or vcc or some
other nearby pin.

imchipwood, 2025-02-05:
Closing issue - no responses in over a year.

closed 2025-02-05 by imchipwood