imchipwood/dumbpad #40: Led question from a newbie

cgdsyldrm, 2023-09-29:
Hello, first and foremost thank you for sharing this beautiful macropad. I tried to find my answer another other issues but I couldn't.

So I am a new to this, both building a custom macropads and soldering but I'm getting there. Is it possible to add those circular led as in the photos? If so, do I need to solder them with a resistor or something? Also do I need to do something with the code?

[image]
[image]

imchipwood, 2023-09-29:
I can't tell which board you're using but none of the boards are set up to use thru-hole LEDs in that way - the routing is very difficult on 2-layer PCBs, and since there's an RGB LED version available, I've never bothered to attempt the routing.

cgdsyldrm, 2023-09-30:
Hello @imchipwood, the board is oled version, I see and thank you for the clarification.

closed 2023-09-30 by cgdsyldrm