imchipwood/dumbpad #41: How to disable running numpad on first/base layer

adrase, 2024-02-14:
Ok, this drive me crazy. What should be done to not have enabled numpad? 
I've got two layers and first one is always replaced with numpad. After I plug keyboard I have arrows, pgup, pgdn, ins etc on first layer but not digits and not what I specified in keymap. Also first led is on. 

Second layer works ok. I work on Ubuntu 22.04
What and where should I change in order to have base layer specified by me and not numpad? Thanks.

Here is my keymap:

#include QMK_KEYBOARD_H

// Tap Dance declarations
//enum {
//    TD_TG_NUM,
//};

// Tap Dance definitions
//tap_dance_action_t tap_dance_actions[] = {
    // Tap once for Escape, twice for Caps Lock
//    [TD_TG_NUM] = ACTION_TAP_DANCE_DOUBLE(TG(1), KC_NUM),
//};

enum custom_layers {
    _BASE,
    _SUB,
};

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
    /*
            BASE LAYER
    /-----------------------------------------------------`
    |             |    7    |    8    |    9    |  /  |
    |             |---------|---------|---------|---------|
    |             |    4    |    5    |    6    |   *  |
    |             |---------|---------|---------|---------|
    |             |    1    |    2    |    3    |   +   |
    |-------------|---------|---------|---------|---------|
    |       mute  |    0    |    .    |    -    |  switch  |
    \-----------------------------------------------------'
    */
    [_BASE] = LAYOUT(KC_P7, KC_P8, KC_P9, KC_PSLS,
                 KC_P4, KC_P5, KC_P6, KC_PAST,
                 KC_P1, KC_P2, KC_P3, KC_PPLS,
        KC_MUTE, KC_P0, KC_PDOT, KC_PMNS, TG(_SUB)
    ),
    /*

    /------------------------------------------------------------------`
    |             |  ctrl s  |   pg up   |   ]          |  wheel up   |
    |             |----------|-----------|--------------|-------------|
    |             |  B       |   pg down |   [          |  wheel down |
    |             |----------|-----------|--------------|-------------|
    |             |   shft i |   H       | shft ctrl a  |  Ins        |
    |-------------|----------|-----------|--------------|-------------|
    |    mute     |   E      |  space    |  ctrl z      |  switch     |
    \-----------------------------------------------------------------'
    */
    [_SUB] = LAYOUT(LCTL(KC_S), KC_PGUP, KC_RBRC, KC_WH_U,
                 KC_B, KC_PGDN, KC_LBRC, KC_WH_D,
                 LSFT(KC_I), KC_H, LCTL(LSFT(KC_A)), KC_INS,
        KC_MUTE, KC_E, KC_SPC, LCTL(KC_Z), TG(_SUB)
    )

};

const uint16_t PROGMEM combo1[] = {KC_P0, KC_PDOT, COMBO_END};
const uint16_t PROGMEM combo2[] = {KC_PDOT, KC_PMNS, COMBO_END};

combo_t key_combos[] = {
    COMBO(combo1, KC_BSPC),
    COMBO(combo2, KC_ENT), 
};

imchipwood, 2024-02-14:
This is more of a question for the QMK community than me... but anyway. What happens when you only define one layer?

Basically, simplify your keymap until you no longer find the issue, then start adding features back in one by one. 

I've never run into an issue like this on a healthy dumbpad.

adrase, 2024-02-15:
Probably you are right I must have messed something. Anyway, I found solution. Maybe not pretty but it works:

`layer_state_t layer_state_set_user(layer_state_t state) {
    switch (get_highest_layer(state)) {
    case _BASE:
        if (!host_keyboard_led_state().num_lock) {
             tap_code16(KC_NUM);
        }
        break;
    default: //  for any other layers, or the default layer
        if (!host_keyboard_led_state().num_lock) {
             tap_code16(KC_NUM);
        }
        break;
    }
  return state;
}`

imchipwood, 2025-02-05:
Closing this since there is a workaround and it's not a dumbpad-specific issue.

closed 2025-02-05 by imchipwood